adafruit/Adafruit-nRF52-Bluefruit-Feather-PCB #3: nRF52840 question: why is the AP2112 present on the board?

lbdroid, 2019-01-16:
The specifications for the nRF52840 include DCDC and LDO power regulators onboard capable of being supplied by any voltage from 1.7 to 5.5, including the ability to drive external 3.3v components (specifically, the multi-color LED and the external flash). The AP2112 has a quiescent current of 55-80 uA, which appears to be an unnecessary waste since the datasheet for the nRF52840 suggest that it can consume as little as 1 uA while being supplied by 5V.

It seems to me that the circuit would be simplified and efficiency improved by removing the AP2112, and instead of tying the nRF52840's VDDH to VDD, connect VDDH to the cathode of D1.

ladyada, 2019-01-16:
we need 300mA minimum for accessories, whats the spec of the nRF52840 LDO?

lbdroid, 2019-01-16:
Ah, ok. Then it makes sense since the nRF is maximum 25 mA. I suppose then that it would be nice to be able to operate without the AP2112 when external accessories are not required.

closed 2019-01-16 by ladyada