adafruit/Adafruit-nRF52-Bluefruit-Feather-PCB #7: Neopixel problems when USB is above 5V (nrf52840)

dsteinman, 2019-06-23:
Although a computer USB connection is usually around 5.0V, some USB chargers put out slightly more than that.  I have 3 on hand at 5.11V, 5.15V, and 5.26V.  All three create problems which in my case, manifest as the first Neopixel on a WS2812 strip flickering green (I am using the Adafruit neopixel library).

The 5.15V one (an iPad Charger) works best of the three, the neopixel still occasionally flickers green, but quite a lot less.  The other chargers are a Samsung phone charger, and a charger that came with a Raspberry Pi 3.

The problems persist regardless of whether the battery is connected.

I noticed in the schematic, that the VBUS line is wired to the nrf pin 32.  I'd need to look into the NRF documentation a bit further to determine what the threshold is for this pin, but I suspect that it may be the cause of my problems.

Would wiring the VHI line to pin 32 be a potential fix?  It's after the D1 diode, so the voltage would be a tad lower, and may resolve the issue.  (I'm probably going to try this out and see the results).

ladyada, 2019-06-23:
hiya you probably need a level shifter - see https://learn.adafruit.com/adafruit-neopixel-uberguide/

closed 2019-06-23 by ladyada

dsteinman, 2019-06-23:
I should have clarified, I am not using the 5V line to power the neopixels.  It is a 24 pixel strip connected to the 3.3V line from the feather board.  So it's already at 3.3V (althought it tests slightly lower ~3.26V).  Everything works good until a USB charger sends >5V.

ladyada, 2019-06-23:
hiya we'd need to see some scope traces of the power/data to see whats going on