adafruit/Adafruit-nRF52-Bluefruit-Feather-PCB #13: Higher power consumption on nrf52840 Rev E boards

donmsmall, 2025-04-24:
Recent [nrf52840](https://www.adafruit.com/product/4062) boards I've bought are now showing an updated PCB version E.  I have not been able to find any documentation concerning changes from Rev D to Rev E. Last revision history on the product page is dated Nov. 2019.   The nrf52840 learning guide contains a reference to [NEOPIXEL_POWER](https://learn.adafruit.com/introducing-the-adafruit-nrf52840-feather?view=all#circuitpython-pins-and-modules) in the CircuitPython section.  I've also found a reference in the Arduino variant.cpp file to a NEO Pixel power pin.  Sounds promissing that we might not need to remove the NEO Pixel LED from the board to get lowerr current.

variant.h

[image]

variant.cpp

[image]

I've been using the Rev D board and removing the NEO Pixel LED to get an idle current of 63uA, which is great for my project.

[image]

But now the Rev E board with NEO Pixel still on the board shows idle current of 1.1mA.

[image]

Do I have the logic backwards to disable the NEO Pixel?  I wrote some test code to prove to myself that setting PIN_NEOPIXEL_POWER high does enable the NEO Pixel and setting PIN_NEOPIXEL_POWER low turns it off.  I've removed the NEO Pixel LED on one board and still get 1.1mA of idle current.  

I've not been able to find a schematic for Rev E, so no idea how to get the current down similar to the discussion in [Issue #4 ](https://github.com/adafruit/Adafruit-nRF52-Bluefruit-Feather-PCB/issues/4).  

Rev E with no mods is 1.1mA and Rev D with no mods is around .5mA.

Any help on power reduction would be appreciated.  Links to schematics would be helpful but I understand how those are probably just copied and your designs stolen so now you have to keep those under lock and key.

ladyada, 2025-05-02:
yes these boards were designed before we spent effort on low power (they were never intended for low power usage :/
@BlitzCityDIY plz post the rev E files at your convenience

BlitzCityDIY, 2025-05-05:
rev E files have been posted

arunshivaram, 2025-06-15:
I recently got a rev E board and was doing power measurements with PPK2 and came across this issue. I was able to get to ~35 μA by just pulling the NEOPIXEL_POWER pin low. Documented the code snippet used and the PPK2 screenshots here in case it is helpful 

https://github.com/arunshivaram/microcontroller_power_profiles/tree/main/Adafruit_nRF52840_Feather_Express

ladyada, 2025-07-04:
@BlitzCityDIY you can update the dox then close

BlitzCityDIY, 2025-07-07:
docs are updated